MaJerle/stm32fxxx-hal-libraries #1: static function declared as non static

GregoireH, 2015-10-30:
Hi MaJerle,

in tm_stm32_usart.c :
I think you missed a static in front of the function definition of
TM_USART_INT_InsertToBuffer(...)

https://github.com/MaJerle/stm32fxxx_hal_libraries/blob/master/00-STM32_LIBRARIES/tm_stm32_usart.c#L81

MaJerle, 2015-10-30:
Hi,

It is not necessary, or you think is it? ;)

GregoireH, 2015-10-30:
well depending on your compiler settings, you get an error or warning... :-)

MaJerle, 2015-10-30:
Whatt??

GregoireH, 2015-10-30:
well, I must confess that :-)
I am using a custom Makefile with sublime 3 text editor,
using arm-none-eabi compiler I got an error saying :
```
 ../00-STM32_LIBRARIES/tm_stm32_usart.c:256:13: error: static declaration of 'TM_USART_INT_InsertToBuffer' follows non-static declaration
 static void TM_USART_INT_InsertToBuffer(TM_BUFFER_t* u, uint8_t c) {
             ^
../00-STM32_LIBRARIES/tm_stm32_usart.c:81:6: note: previous declaration of 'TM_USART_INT_InsertToBuffer' was here
 void TM_USART_INT_InsertToBuffer(TM_BUFFER_t* u, uint8_t c);
      ^
Makefile:122: recipe for target `../00-STM32_LIBRARIES/tm_stm32_usart.o' failed
```

MaJerle, 2015-10-30:
Aaa this.  Ok yeah! Will fix it.

GregoireH, 2015-10-30:
Great,

I ll email you for a personal question then.
Cheers

closed 2015-12-18 by MaJerle