MaJerle/stm32fxxx-hal-libraries #2: ADC Library exceeding maximum code size

anirudt, 2015-10-31:
Loading project number 29 onto Keil uVision 5, followed by compiling and linking yields the following error message. 

```
.\Targets\STM32F7_DISCOVERY\project.axf: error: L6050U: The code size of this image (38088 bytes) exceeds the maximum allowed for this version of the linker.
Finished: 0 information, 0 warning, 0 error and 1 fatal error messages.
".\Targets\STM32F7_DISCOVERY\project.axf" - 1 Error(s), 0 Warning(s).
```

A similar issue is found with the DAC library pack. Is there any trivial fix for the above issues with the free software version of Keil uVision?

MaJerle, 2015-10-31:
Remove unused files from build or buy license.

anirudt, 2015-10-31:
Could you be a little more specific about "unused" files, pertaining to the application? I ran the DAC application, freshly cloned from the repo. Also, I observed that it was getting compiled for STM32F4** targets. Could you check it for STM32F7?

closed 2015-12-18 by MaJerle