MaJerle/stm32fxxx-hal-libraries #3: ADC Library STM32F030x6 Support

burakenez, 2015-11-16:
ADC library gives error for STM32F030x6 devices. In order to avoid this error,
void TM_ADC_EnableVbat(void);
void TM_ADC_DisableVbat(void);
uint16_t TM_ADC_ReadVbat(ADC_TypeDef* ADCx);
functions may be blocked out because STM32F030x6 doesn't have vbat peripheral and registers.

MaJerle, 2015-11-16:
Hi, 
I didn't check that and this may be true. Will strip them out in that case.
What exactly was an error you received?

burakenez, 2015-11-16:
compiling tm_stm32_adc.c...
..\..\Libraries\tm-stm32lib\tm_stm32_adc.c(177): error:  #20: identifier "ADC_CCR_VBATE" is undefined
  	
DC->CCR |= ADC_CCR_VBATE;
..\..\Libraries\tm-stm32lib\tm_stm32_adc.c(186): error:  #20: identifier "ADC_CCR_VBATE" is undefined
  	
DC->CCR &= ~ADC_CCR_VBATE;
..\..\Libraries\tm-stm32lib\tm_stm32_adc.c(194): error:  #20: identifier "ADC_CHANNEL_VBAT" is undefined
  	result = TM_ADC_Read(ADCx, (TM_ADC_Channel_t) ADC_CHANNEL_VBAT);
..\..\Libraries\tm-stm32lib\tm_stm32_adc.c: 0 warnings, 3 errors

closed 2015-11-16 by burakenez