MaJerle/stm32fxxx-hal-libraries #5: TM_RTC_Config

qermit, 2015-11-26:
TM_RTC_Config does not set all RCC_PeriphCLKInitTypeDef struct entries.
PeriphClockSelection may have random value and HAL_RCCEx_PeriphCLKConfig will hang.

MaJerle, 2015-11-26:
Let me know about yourfix. According to the clock you wanna init, these
options are used and other ignored.

Waiting your fix.
Have a nice day.

qermit, 2015-11-26:
I've created pull request #6 
I'm not sure if this is correct (works with stm32f429l-discovery).

qermit, 2015-11-26:
> Edit: I have check your "bug" post. As you can see from HAL_RCCEx_PeriphCLKConfig function, it first check for which clock TYPE it goes. If RTCCLK is selected, then it does not use any other types of structure, only these.

**HAL_RCCEx_PeriphCLKConfig** is correct, however nothing is selected in **TM_RTC_Config** and **RCC_PeriphCLKInitTypeDef.PeriphClockSelection** has some garbage that was previously in this memory area. So in my case (24-STM32Fxxx_RTC_LCD) **RCC_PeriphCLKInitTypeDef.PeriphClockSelection** was equal to 1

[image]

MaJerle, 2015-11-26:
Yes, true that. I've fixed it, just must upload code.

Thanks for that.

closed 2015-11-26 by MaJerle