MaJerle/stm32fxxx-hal-libraries #13: Receive data error

dinhdiep9947, 2016-03-29:
Hi Majerle,
I use kit STM32-nucleo F411 with your library to transmit and receive data on nRF24L01. In the NRF24L01P_RECEIVER project, I debug program, variable `dataIn` get incorrect data despite no transmition data.

MaJerle, 2016-03-29:
Hi, 

Do you have solution to this maybe also?

dinhdiep9947, 2016-03-29:
Hi,
Really I don't good about this. I try to understand your code. And I want to use your library combine with CubeMX for new programer like me may program STM more easy.

MaJerle, 2016-03-29:
Yeah I uderstand. But as far I know, library worked ok but anyway, if I find time, I will check on it to see what can go wrong. For me always worked.

dinhdiep9947, 2016-03-29:
Thanks your reply. I port your command to HAL command.
`TM_SPI_Send(SPIx, data)` -> `HAL_SPI_Transmit(SPIx, data, 1, 1000)`
`TM_SPI_SendMulti(SPIx, dataOut, dataIn, count)` -> 'HAL_SPI_TransmitReceive(SPIx, dataOut, dataIn, count, 1000)'
'TM_SPI_WriteMulti( SPIx, dataOut, count)' -> `HAL_SPI_Transmit(SPIx, dataOut, count, 1000)`
'TM_SPI_ReadMulti(SPIx, dataIn, dummy, count)' -> `HAL_SPI_TransmitReceiver(SPIx, dataIn, dummy, count, 1000)`
So what do U think? This's true?

MaJerle, 2016-03-29:
Why would you do that?

dinhdiep9947, 2016-03-29:
When CubeMX genarate code, I want these library can use. If I include your library, I wasn't use to CubeMX to genarate. Maybe I still do not good enough to use both.

MaJerle, 2016-03-29:
Not sure I understand you.

dinhdiep9947, 2016-03-29:
I speak English very bad :(

dinhdiep9947, 2016-03-29:
I use your library, NRF24L01P_TRANSMITTER project,`transmisionStatus = TM_NRF24L01_Transmit_Status_Sending` always. I don't know what I wrong :(

closed 2016-09-08 by MaJerle