MaJerle/stm32fxxx-hal-libraries #14: Data via mobile network

suhasbhairav, 2016-04-21:
Is it possible to transfer data using GSM2 click module on a discovery board using the HAL libraries? I was not able to find any library associated with GSM/GPRS to do the same.

MaJerle, 2016-04-21:
it is possible.
You need USART library to interface your GSM module.

I don't have similar library like my esp8266at.com but I'm planning to make similar like that for SIM800 or SIM900 devices.

suhasbhairav, 2016-04-21:
So does the USART library work for GSM 800/900 ?

MaJerle, 2016-04-21:
I don't know what knowledge you have in microcontrollers or similar, but:

1. USART is only for communication. Sending/Receiving data.
2. You have to know by yourself what you will send and what you wanna receive back.

closed 2016-09-08 by MaJerle