MaJerle/stm32fxxx-hal-libraries #26: Unstable FatFS SPI SD Card Write

enochtsang, 2018-03-07:
# Background

I'm using the FatFS HAL driver with the STM32F405RGTx MCU on a custom PCB. I'm with the University of Calgary's Student Organization for Aerospace Research, and we are using the STM32 for our onboard flight systems for our rocket that will compete in the Spaceport America Cup 2018.

I followed the guide on your website at http://stm32f4-discovery.net/2015/08/hal-library-20-fatfs-for-stm32fxxx/

# Final Result

I did eventually get the MCU to write to the SD card, to `first_file.txt` with the correct information (I was so happy). But it was very very unstable. With a myriad of power cycles and reflashing the MCU, seemingly at random, sometimes the card will get mounted and the card will get written to, and otherwise it would not get mounted. 

If the SD card gets mounted, it happens around 3 seconds after startup. Otherwise the SD card never gets mounted (I've waited up to a few minutes). If the SD card gets mounted, it always gets successfully written to.

## Code Changes

I downloaded the code located on your website, which I did notice was out of date. I had to implement the fix on #25. I did some other changes as well so unnecessary includes didn't mess up the build.

Once the build worked, the mount was failing every time, it boiled down to the `select()` function in `fatfs_sd.c` on line 158. The timeout number was too short `if (wait_ready(500)) `, I changed it to `10000`, and then sometimes the disk would get mounted. It takes about ~3 seconds every time it works.

## My Code 

This is the code I used to write to the SD card.
https://github.com/enochtsang/STM32F405RGT6_SdSpiWriteExample

These are the defines I used in my `defines.h` file.
``` c
#define STM32F4xx
#define FATFS_USE_SDIO 0
#define FATFS_SPI SPI3
#define FATFS_SPI_PINSPACK TM_SPI_PinsPack_2
#define FATFS_CS_PORT GPIOD
#define FATFS_CS_PIN GPIO_PIN_2
```
And the initialization parameters for the SPI peripheral.
```c
  hspi3.Instance = SPI3;
  hspi3.Init.Mode = SPI_MODE_MASTER;
  hspi3.Init.Direction = SPI_DIRECTION_2LINES;
  hspi3.Init.DataSize = SPI_DATASIZE_8BIT;
  hspi3.Init.CLKPolarity = SPI_POLARITY_LOW;
  hspi3.Init.CLKPhase = SPI_PHASE_1EDGE;
  hspi3.Init.NSS = SPI_NSS_SOFT;
  hspi3.Init.BaudRatePrescaler = SPI_BAUDRATEPRESCALER_4;
  hspi3.Init.FirstBit = SPI_FIRSTBIT_MSB;
  hspi3.Init.TIMode = SPI_TIMODE_DISABLE;
  hspi3.Init.CRCCalculation = SPI_CRCCALCULATION_DISABLE;
  hspi3.Init.CRCPolynomial = 10;
```

# Questions

So with all that, I have two things I would like to ask.

1. **Is increasing the timeout a change suitable to this repo? I can see how it might not because FatFS is a third party library. But if it is, I'd happily make a pull request for it.**

2. **Based on my setup, is there any red flags for instability with the SD card? Do you have any ideas for how to increase the stability of writing to the SD card?**

MaJerle, 2018-03-07:
DId you test this with single or different SD cards?

enochtsang, 2018-03-09:
You were correct, the SD card faulty. It ran fine doing IO on a computer though.

closed 2018-03-09 by enochtsang