MaJerle/stm32fxxx-hal-libraries #35: USART showing garbage

lekoook, 2019-08-03:
I've tried using the USART library and running the example in 07-STM32Fxxx_USART.
However, I am getting garbage values on the terminal.
I've tried both putty serial on Windows 8.1 and minicom on Ubuntu 18.04. Both are configured at 921600 baud (I've also tried 9600, 115200), 8 data bits, 1 stop bits, None parity.
I've also tried these two USB to TTL: https://shopee.sg/product/40263913/709423835 & https://shopee.sg/product/12694316/1274615610

Any idea what could be the issue? Any configurations I am missing or wrong hardware used?

MaJerle, 2019-08-03:
Might be related with system clock setup.

lekoook, 2019-08-03:
Thanks for pointing in the right direction!

However, TM_RCC_InitSystem() seems to be broken and interestingly, by removing that function, the USART printing works perfectly fine. Why is that so?

Also, this library appears to only print purely characters and strings. So I guess I will have to do my own implementation for printing ints, uints and doubles right?

lekoook, 2019-08-03:
I've noticed TM_RCC_InitSystem() sets the SystemCoreClock variable to 562500000. Is that normal?

MaJerle, 2019-08-03:
No it is not normal. Reason is hse clock. I assume you have it set to 25mhz for some reason and it should be 8. 25/8 is similar to 525/180 mhz if you use F4.

Be advised that this repository is for eval purposes, and not all features are implemented. It is not framework.

closed 2019-08-03 by MaJerle