MaJerle/stm32fxxx-hal-libraries #39: CubeMX IDE, how to configure and use the library in a project

lafar6502, 2020-03-19:
Hi, so i'm trying to make use of your libraries in a stm32 F4 project in stm32 CubeMx IDE. And i'm not sure how it's supposed to be done
1. Should i include / compile the whole library or just pick single files? Would prefer the whole
2. Is it possible to use git submodule or somehow check out hal libraries folder from git, independent of my own repository, without copying all your library into my project? Best way of doing that?
3. Setting up include folders and include files in order to get the full library to compile. Which include folders should be added to compiler search list? What header files need to be added by me - config.h? check.h (not sure about it but seems missing)? other?
you know, just a best practice how to use it without going thru too much mess and reinventing everything. Thanks!

lafar6502, 2020-03-19:
update - trying to get the library to compile (with stm32 F401 discovery board) in various ways, but something is always missing. Now i'm going with minimal file set and got stuck on SPI_FLAG_TXE definition missing. It's nowhere to be found in my header files, the only place is Driver/Inc/Legacy/stm32_hal_legacy.h file , but it's only available for STM32H7 board. Please, some suggestions how to use..

MaJerle, 2020-03-20:
> 1. Should i include / compile the whole library or just pick single files? Would prefer the whole

There are many libraries, not just one, so you need to decide first which one you are interested in.

> 2. Is it possible to use git submodule or somehow check out hal libraries folder from git, independent of my own repository, without copying all your library into my project? Best way of doing that?

Git does not support cloning only single folder. You have to clone entire repo.

> 3. Setting up include folders and include files in order to get the full library to compile. Which include folders should be added to compiler search list? What header files need to be added by me - config.h? check.h (not sure about it but seems missing)? other?
>    you know, just a best practice how to use it without going thru too much mess and reinventing everything. Thanks!
>    asking for help because now i'm getting errors trying to compile the whole lib, especially in the subfolders (lwip) - headers missing, unit test library missing, etc..

You need to put to include all paths where header files are. But it makes no sense to compile everything if you are not using it.

closed 2020-04-01 by MaJerle